Heatit-Controls/no.ThermoFloor #1: zwave pairing

MenneBos, 2018-05-19:
Hey,

I just bought a two heatit fr021 and tried to connect them to the homey. I cannot get them paired with the Homey. I followed the instructions and changed the following:
- one meter distance
- reset zwave network
- disconnect heatit device

Is there something I can do on the Homey?

Tanks in advance, for any response, regards Menne

TedTolboom, 2018-05-22:
Hi @MenneBos, 

Let's see if we can solve it..
Can you indicate which Homey SW release you are working on?

Can you try the following steps:
To remove the thermostat from Homey (eventhough you think it was not included): 
1. press the Center (confirm) button for 10 seconds. The display will show OFF. 
2. Press Right(down) button 5 times till you see Con in the display.
3. Now start to remove the device from Homey (Go to Settings > Z-wave > remove a device).
4. Start exclusion mode by pressing Center(confirm) button for approximately 2 seconds. The inclusion / exclusion icon will appear in the display.
5. Confirmation will show EcL in the display. If exclusion fails, Err(error) will appear.

Reboot Homey and wait at least 15 minutes to allow all apps to be loaded completely.

To include the thermostat to Homey: 
1. press the Center (confirm) button for 10 seconds. The display will show OFF. 
2. Press Right(down) button 5 times till you see Con in the display.
3. Now start add a device to Homey (devices > + > ThermoFloor App > select TF thermostat).
4. Start inclusion mode by pressing Center(confirm) button for approximately 2 seconds. The inclusion / exclusion icon will appear in the display.
5. Confirmation will show Inc in the display. If inclusion fails, Err(error) will appear.

Can you indicate if any error is shown during exclusion / inclusion?

MenneBos, 2018-05-25:
Tx Ted,

I will execute the removal  procedure (eventhou I cannot see any heatit device connected) and restart procedure as soon as possible and I will let you know the results.

Homey version 1.5.11

Tx again Menne

Verstuurd vanaf mijn iPhone

ParkerJulius, 2018-08-12:
Hi!
I have a similar problem, I was able to add one thermostat, but when I tried adding the rest everyone of them just says err and Homey says "no wave device found". tried including it to another controller I have and that is working fine.

ParkerJulius, 2018-08-12:
Homey Version: 1.5.11
Thermo-Floor AS v1.0.0

TedTolboom, 2018-08-12:
At what distance of Homey are you trying to include the thermostat?

ParkerJulius, 2018-08-12:
That was it! Worked fine now!
Was trying at 10-15 meters before.
Thanks ! 🙏

TedTolboom, 2018-08-12:
Great, thanks for the feedback!
Will close this issue, since original last feedback was from May this year.

closed 2018-08-12 by TedTolboom

TedTolboom, 2018-08-14:
@MenneBos Thanks for the feedback and response!