Heatit-Controls/no.ThermoFloor #2: Z-Wather

edorf83, 2018-09-16:
When can I use Z-Water in Homey?  I have alredy bougth this device, but I cant use it yet..

MbenzNo, 2018-09-19:
Hi! any plans for adding other HeatIt/Thermo floor devices to this app? Would like to have support for the 25A relay switch : https://www.tronika.no/no/index.php?controller=attachment&id_attachment=682

TedTolboom, 2018-10-07:
See my comment at the forum:
https://community.athom.com/t/thermofloor-multireg-heat-it-app-v1-0-0/166/15

TedTolboom, 2019-01-04:
@edorf83 @MbenzNo Brief update on the Z-water and Z-relay.
I just released the v2.1.0 update that includes the Z-TRM2(fx) support to the app store.

Pending feedback on that release, I will start working on implementing the Z-relay and Z-water devices.

TedTolboom, 2019-02-10:
Initial support for the Z-water is added in v2.1.1 (BETA)

Temperature readout is still not possible due to a bug in the Z-wave core.
All 10 relays can be controlled.

closed 2019-02-10 by TedTolboom