Heatit-Controls/no.ThermoFloor #3: HEATIT Z-TRM2

lthierry, 2018-09-25:
Hi Ted,

I am considering to buy the new HEATIT Z-TRM2:
https://www.heatit.com/heating-control/floor-heating-thermostats/heatit-z-trm2

This new version of the HeatIt thermostat includes over-the-air updates and some other new features.

Is this product already supported by your Homey app? 
If not, would you be willing to add it soon?

Kind regards/
vriendelijke groet,
Leonard Thierry

SenseiJanuZ, 2018-09-26:
It's not. I've got the HEATIT Z-TRM2 and what happens is that the thermostat get's included but then Homey gives the message that it doesn't recognize it so it's seen as a unknown Z-wave device that can be turned on or off ...

lthierry, 2018-09-26:
Thanks.
Ted, are you willing to add the HEATIT Z-TRM2?

TedTolboom, 2018-10-07:
See my comment at the forum:
https://community.athom.com/t/thermofloor-multireg-heat-it-app-v1-0-0/166/15

TedTolboom, 2019-01-04:
Z-TRM2(fx) support is added with v2.1.0 and will be released as Beta release to the app store (awaiting Athom's approval).

I will close this issue. Please open a new issue if issues / feature requests originate from v2.1.0

closed 2019-01-04 by TedTolboom

lthierry, 2019-01-04:
Thanks very much, Ted.

Looking forward to use the update for my 5 Z-TRM2 thermostats.

In particular, I hope to be able to use an external temperature sensor in
my Z-wave network, a few meters away from the thermostat, to do the
temperature measurement for the Z-TRM2 instead of using the unreliable
internal temperature sensor or the standard external temperature sensor
that is getting too much heat from the thermostat, because it's in the same
wall mounting box.

Kind regards,
Leonard

SenseiJanuZ, 2019-01-04:
Thx Ted!

It's guys like you that make Homey what it is!

Can't wait to use it.

TedTolboom, 2019-01-04:
@lthierry I'm not sure if that is possible with the Z-TRM2 device itself.
From Z-wave perspective, it doesn't accept external temperature readings.
You can use Homey flows to control the thermostat, toggling the thermostat mode between off and comfort mode based on an external sensor; setting the setpoint of the comfort mode to a certain value. Otherwise, let's follow-up in the Homey community forum topic.

@SenseiJanuZ Thanks, your feedback is much appreciated.
Sometimes people underestimate the amount of effort required to develop these apps (and support them). ;-P

SenseiJanuZ, 2019-01-04:
@TedTolboom..  Support will follow ;-)