Heatit-Controls/no.ThermoFloor #7: Parameter 12 as an action card (ztrm2/ztrm2fx)

henrikrox, 2018-11-25:
Hi 

So,

I guess this goes for both ztrm1 (multireg) and ztrm2/ztrm2fx. The interal sensor on this thermostats arent really that great. When i tried out OpenHab i saw a great little fix for this. Now Thermofloor doesnt allow external sensors to control the thermostats. 

But in P mode it does, if Parameter 12 is made available in a flow.
[image]

In this case, setting this paramater to 10, would mean that the heating would stay on all the times, and a value of 0, means always off. I would advise people (if this comes available) to have some sort of failsafe in a flow. Because wireless sensors could fail (loss of batter and so on). Especially when it comes to heating on hardwood floors.

But back to subject, this is especially helpful for when the thermostat is mounted on the outside of the bathroom, and you dont have a floor sensor. A quick fix, is to use a wireless zigbee or zwave temperature device inside the bathroom, set ztrm1 or 2 in P mode, and set Paramater 12 to on or off based on read temperature from the sensor inside the bathroom.

A typical scenario would be,

So if the setpoint temperature has been reached, run parameter 12 value = 0
If the temperature is lower then the setpoint, run paramater 12 value = 10

- henrik

TedTolboom, 2018-12-30:
Hi Henrik,

On which firmware are you with the Z-wave thermostat?
I've added the settings and flow card for the FW 1.92 thermostat at the Z-TRM2fx branch: https://github.com/TedTolboom/no.ThermoFloor/tree/Z-TRM2fx

TedTolboom, 2019-01-04:
@henrikrox with the 2.1.0 (at first in Beta) these action cards will become available.

I'll close this issue for now. Please re-open if you original request is not fullfilled

closed 2019-01-04 by TedTolboom