Heatit-Controls/no.ThermoFloor #12: Heatit Z-TRM2fx inclusion ok, failed afterwards

maja2020, 2019-02-10:
You asked for an issue in the apps forum. 

I have everspring, fibaro and eurotronic devices in the Z-wave network. These work OK. 
I included the  Z-TRM2fx, inclusion was difficult, the therm has a short time-out period. 
I did get it included. It did work for an hour or so, after that, the zwave network became unresponsive, 

I tried to exclude, that went wrong. 
I now have a situation where the zwave device is an unkown device, but it is communicating with homey on the zwave network. Homey app does not see the device. I have more "unkown devices", leftovers from re-inclusion. This is a know issue, but no big deal since the devices do not communicate any more. Counters stay at "-" after a homey reboot. 

My therm  Z-TRM2fx is an unkown device but communicating.  
Reinclusion does not work: the app says "device is already included". Unable to delete through zwave developer tools and the apps zwave settings page. I sent a support ticket to athom to help me delete the node so I can try re-including the device.

The heatit therm is node 83. It shows counters, 20 min. z-wave log for node 83: empty. 
When I can reinclude agin, I can send you z-wave logs.
[image]

maja2020, 2019-02-13:
Athom reported back from my issue report. Athom says I should try contact the app developer, nothing wrong with my Homey. 

Yesterday I disconnected all zwave devices from the network and performed a z-wave-reset (new zwave network). I started to pair all the devices (12), The factory resetted heatit device first. I tried to include with the app and the homey "zwave device connection" device card, at least 10 times. 
Did not work. 

All other devices paired properly. 

Today, I disconnected the heatit therm from power, to force a hard reset of zwave chips on the device that may have fallen in an uncontrollable state. The results are the same: the device seems "z-wave" dead. 

The only possible explanation right now is that the zwave function of the therm is broken. 
That may also explain the initial strange behaviour. 

I am going to ask for a replacement therm. If that works, I am sorry for an unneccesary open issue. If it does not, I will report back. Thx.

maja2020, 2019-03-09:
I replaced the therm with a new one. I was able to pair the device successfully. After a homey reboot (the 2.0.5-rc.4 update), the device stopped functioning. In the z-wave dev tools page it showed up as "uknown" device. 
I tried to re-pair the device witout and with a factory reset. I also took homey to the device to exclude z-wave comm problems. The results are just the same: device ends in erro, the app in a "no z-wave device detected". 

So I decided to make a log trace of the pairing process. With my limited knowledge of the z-wave protocol. It seems the device and homey communicate, but only command_class_security  messages. 

I hope you can take a look at it and help figure out what's wrong. 

[pairinglogHeatit.txt.txt](https://github.com/TedTolboom/no.ThermoFloor/files/2948365/pairinglogHeatit.txt.txt)

TedTolboom, 2019-03-13:
@maja2020 can you try the following steps with Homey in close proximity of the thermostat:
1) reset the Z-TRM2fx by holding the right two buttons for at least 20 seconds until RES is shown on the display; it will restart and show the device firmware (3.4)
2) on the device, go to the menu, INC and wait
3) on Homey app, go to settings > Z-wave and remove Z-wave device
4) now activate the Inclusion mode on the thermostat

Now the device should be reset and removed from Homey.

Now try to include it again; normal steps

Can you let me know if that fixes the issues?

maja2020, 2019-03-13:
I can confirm the firmware version is 3.4.

I finally managed to include the device. 
I had to include, remove the device from homey (the inclusion timed out -> unknown z-wave device). Tried again and then it worked. 
The differences: 
- homey version 2.5 
- I did not first configure the device manually after the factory reset. 
The temp NTC is a 12kohm version. That needs to be set to the correct value, otherwise the floor temp sensor reports wrong values. I set the value through the general settings of the app.

closed 2019-03-13 by maja2020