Heatit-Controls/no.ThermoFloor #13: Missing data in insights

runesyd, 2019-03-03:
I have 3 floor thermostats. None of them shows anything in insights, but they are working as expected in the app on my iphone.

It's this type:
https://www.heatit.com/heating-control/floor-heating-thermostats/heatit-z-wave-thermostat/

I've been in touch with homey support who said I should get in touch here.
Please let me know if it is anything I can do to help investigate the issue.

olemart, 2019-03-27:
I'm also missing data in insights. I have 3 heatit z-trm2fx thermostats. They worked initially, inclusive wattage report, however after a few days they stopped reporting. They floors are still warm, so that part works. In insights under "power" on two of them, it shows no further wattage consumption after they stopped, (not simultaneously). On the third thermostat it shows continuously 633 watt. In insights under "power meter" they all stopped accumulating kwh after they stopped. I tried to exclude all three, factory reset all z-trm2fx, include again, and the same thing happened again. It seems like everything works perfect for a couple of days, and then they stop. If I check in the Homey app on the specific device, turn up the thermostat a lot, nothing happens with "power", I also cannot see floor temperature. How can i help finding out whats wrong? (I'm no programmer, so don't know what is needed).

TedTolboom, 2019-09-18:
@runesyd is this still an issue for your thermostat?

@olemart please try v2.3.0-BETA has been uploaded to the Homey Apps store to the BETA branch of the Heatit app, awaiting Athom's approval. Could you try to remove the Z-TRM2fx and re-include it based on this version?
This update should resolve stability issues that could prevent temperature readings to be processed

olemart, 2019-10-08:
Hi. Sorry for the delayed response. Seems like my problem is something differently. I found the homey developer page a while ago (developer.athom.com, for others that might find this post), where I saw that all three Heatit thermostats was trying to connect through a nonexisting node. I was not able to heal it, and I had to factory reset the heatit, remove the connection from homey, and reinclude. I now got the same problem with just one of the heatit thermostats, it is now trying to connect through node 16 which does not exist on my network, and no values have been updated (but also no errors anywhere). I have not included or excluded other products in this time. My other two heatit thermostats are still working great now, for about two months now. I will write to homey so that they can look into this. 
Thanks for all your effort being put into this app! Keep up the good work! I will update to the latest betaversion now, and inform you if anything changes.

olemart, 2020-01-08:
Hi. It seems like my problem returned with no info in insights. One of my heatit z-trm2fx thermostats shows the same "Power" and "Power meter" since november 4th. The floor still gets warm, so took a while before I noticed this. When I look at Homey Developer, I see that it tries to connect through a non-existing node nr 16. When I click "test", it shows "Node 40 is reachable". 
On my network there's also another z-wave product (product: Pulse. App: Tibber), measuring energy consumption for the whole house, which also do not work, and tries to connect through node 16. Since this has nothing to do with your app, I suspect the problem does not lie within your app. I will contact athom support, the last time they said it was an app not developed by them and therefore they will not help, let's see now. I also have problems connecting another product where you are the author, ID Lock 150 where inclusion does not work, no idea if this is related, I wrote a different thread about this. If there's any information I can give if you suspect something, please let me know.