Heatit-Controls/no.ThermoFloor #14: status.already_added

mhauken, 2019-03-24:
Tried to include it, but it seemed to stop. Then when I tried adding it again it says `status.already_added` in Homey-app and err after a while on the termostat. I have the multireg with three buttons under the display. Worked fine earlier before I excluded it from futurehome

mhauken, 2019-03-24:
Added a bunch of thermostats today. Mostly worked, but got the same issue with another one. Seems like it stopped pairing during `Retrieving command class versions (5/12).`. It stopped on that message, then I got an error. When I try tp pair again I get the same `status.already_added`

TedTolboom, 2019-09-18:
@mhauken  v2.3.0-BETA has been uploaded to the Homey Apps store to the BETA branch of the Heatit app, awaiting Athom's approval. Could you try to re-add the thermostat based on this version?