Heatit-Controls/no.ThermoFloor #17: Heatit Z-TRM2fx Showing temperature

GunnarEM, 2019-06-13:
Hi Ted,
I just got 8 Heatit Z-TRM2fx thermostats installed. Two of them does not communicate the floor sensor temperature to Homey. The others are working fine. Do you have any suggestions for how to rectify this?
Thanks!
Gunnar

TedTolboom, 2019-09-13:
Hi @GunnarEM,

Appologies for the late response. I've been out for an extended period.

Have the issues with the two thermostats not reporting the floor temperature been resolved?
Can it be that these have FW 3.6 installed?

TedTolboom, 2019-09-15:
@GunnarEM finally found some time this weekend to dig into the issues with the Z-TRM2fx.
It appears to be related to S2 security handshakes going bad leading to excessive communication clogging the network / Z-wave chip. See the post on the forum for more details: https://community.athom.com/t/heatit-z-wave-app-v2-2-1/166/173

TedTolboom, 2019-09-18:
@GunnarEM  v2.3.0-BETA has been uploaded to the Homey Apps store to the BETA branch of the Heatit app, awaiting Athom's approval. Could you try to remove the Z-TRM2fx and re-include it based on this version?

GunnarEM, 2019-09-22:
Thank you for your response on this Ted. They seems to work fine with the new beta update.

TedTolboom, 2019-09-22:
Thanks for the feedback; I’ll close this issue.
If new issues pop-up, please create a new issue.

closed 2019-09-22 by TedTolboom