Heatit-Controls/no.ThermoFloor #19: Z-water not responding

peskareta, 2019-07-21:
One of my two z-waters do not respond. I get the red message with 30000ms warning. Is it necessary to re include the z-water after thermofloor update? The relays are 2meter away from the Homey.

TedTolboom, 2019-09-15:
@peskareta appologies for the late reply.
I'm looking into the issues with the Z-water and Z-TRM2fx which might be related to a S2 security handshake leading to excessive communication that is clogging the network / Z-wave chip.

Please see the following community post for more details: https://community.athom.com/t/heatit-z-wave-app-v2-2-1/166/173

TedTolboom, 2019-09-18:
@peskareta  v2.3.0-BETA has been uploaded to the Homey Apps store to the BETA branch of the Heatit app, awaiting Athom's approval. Could you try to remove the Z-Water and re-include it based on this version?