Heatit-Controls/no.ThermoFloor #21: Z-relay not reading Temp sensor 10k ntc

kjetilknutsen73, 2019-08-11:
Hi 
Have tried 3 different 10k NTC Temp scensor but i dont get any Temp reading ..

Settings = 10k Ntc

TedTolboom, 2019-09-18:
@kjetilknutsen73 v2.3.0-BETA has been uploaded to the Homey Apps store to the BETA branch of the Heatit app, awaiting Athom's approval. Could you try to remove the Z-Relay and re-include it based on this version.
This update should resolve stability issues that could prevent temperature readings to be processed

kjetilknutsen73, 2019-10-10:
Works GREAT !!!!!!!!!!!!!!!!!!!!!!!!!

Thx

Kjetil Knutsen
Leder IT-støtte

+47 51 87 50 54
+47 901 04 505

Prof. Olav Hanssensvei 15, 4021 Stavanger, Norway

[cid:norce_426cc920-64ab-4893-95d0-c29d5dab670c.png]

NORCE Norwegian Research Centre AS
norceresearch.no<https://www.norceresearch.no/>

TedTolboom, 2019-10-11:
Thanks for the feedback. I'll close the issue then.

closed 2019-10-11 by TedTolboom

kjetilknutsen73, 2019-10-18:
Hi

Just one thing.

Dies not display voltage

[cid:5F253A82-3A40-4B48-A083-51E01A13B418]

Mvh
Kjetil Knutsen

Kjetil Knutsen
Leder IT-støtte

+47 51 87 50 54
+47 901 04 505

Prof. Olav Hanssensvei 15, 4021 Stavanger, Norway

[cid:norce_426cc920-64ab-4893-95d0-c29d5dab670c.png]

NORCE Norwegian Research Centre AS
norceresearch.no<https://www.norceresearch.no/>