Heatit-Controls/no.ThermoFloor #28: Getting instruction on how to remove heatit, when trying to add it

catohauk, 2019-12-16:
Hi.

I'm trying to add the Z-wave thermostat (not the 1.92v). When doing so, the instructions only explain how I can remove the thermostat, not how to add it. 

Please advise on how to add the device. 

[image]

Best regards,
Cato

TedTolboom, 2019-12-16:
@catohauk this is apperantly a non communicated change on Homey firmware side; to force the removal of a device first (step 1, in above screenshot) and then include it (step 2).
I also learned this yesterday, while debugging a device.

Just follow the instructions shown, first trigger the removal, then the screen will change to the inclusion instructions (almost the same) and the device will be included.

catohauk, 2019-12-16:
Allright. I'll try that. Thanks Ted

RAN-dale, 2020-02-26:
I have the same problem. I have tryed 100 times, and the same is happening. I also have Heatit with extern sensor. I got the same instrution that explain how too remove. It also shut down/ reset and err4. I have changed settings to ekstern sensor and have too set the settings again...

Madaxhub, 2020-07-05:
I have the same problem. Could you please update the app 😊

HeatitControls, 2021-06-30:
@RAN-dale step 1 of inclusion is removing the device and after you've proceeded with that it will show you how to include it.

closed 2021-07-30 by HeatitControls