Heatit-Controls/no.ThermoFloor #31: Source-code changes reverting to default when installing to homey

hendrikjansen, 2019-12-23:
I've downloaded the source and added an extra line in the app.json file:
**"includeSecure": false**
in the "Z-push-button-8"-section to force the inclusion to be UNsecure. Otherwise it will not associate with the Z-Dim that is also added UNsecure.

I saved the file and then did a:
**homey app install** 
in the NodeJS command-prompt.
This installs the app in my homey. 

But unfortunatly every time I do the install the source gets reverted to the "original" version, so without my change. 

I can see it happening in my text-editor, that warns me: "The file has been modified by another process, reloading"

So the NodeJS is actually un-doing my changes. Preventing me from using the altered version of the code.

Any idea why this is happening?

TedTolboom, 2019-12-23:
You didn’t sign up for the monthly subscription fee to use the premium feature, didn’t you? 
Then the protection mechanism is not disabled and overwrites all changes...

TedTolboom, 2019-12-23:
Sorry, couldn’t resist.
While installing the app, the app (and in particular it’s manifest app.json) is composed out of the refactored components. 
So if you make changes in the driver section of app.json, the changes are overwritten by the refactored components.

You should apply the changes in drivers/Z-Push-button-8/driver.compose.json and then it will work.

TedTolboom, 2019-12-23:
Happy X-mad holidays!

hendrikjansen, 2019-12-24:
Great! That worked and fixed my problem. I'd like to enroll on the monthly subscription now! :)

closed 2019-12-24 by hendrikjansen