Heatit-Controls/no.ThermoFloor #34: No temperature data z-water

2tracer, 2020-01-27:
Hello.
I do not get any sensor to show temperature in the z-water (relay 1) on the Homey.
I have connected 4 10k ntc sensors.
Do you know how to fix this?

Best regards
Karl

2tracer, 2020-01-27:
[image]

2tracer, 2020-01-27:
I asked heatit on facebook messenger and the answered me this:
You must put assosiations from endpoint 0 (group 1) to gateway with multichannel encapsulation.

Do you know what they mean? 😇😇

TedTolboom, 2020-01-27:
They are originally set in this way during the inclusion process; is it possible to send me the diagnostics report and a screenshot of the advanced settings (association groups)?

2tracer, 2020-01-27:
Hello.
Yes offcorse. :-)

I named the report, Report missing sesor data:
61fdeadc-391a-4f6d-ad5c-0327ade26d13

[image1.png]

[image2.png]

[image3.png]

Best regards

Karl

Sendt fra min iPhone

2tracer, 2020-01-28:
[image]
[image]
[image]

2tracer, 2020-01-28:
Did i do it rigth? :-)

2tracer, 2020-02-01:
Hey Ted. Did you get what you need? Best regards

2tracer, 2020-02-07:
Hello.
I fikses this by changing the paramerer on assosiation. Now it works good. 👍👍

[image]

closed 2020-02-07 by 2tracer

TedTolboom, 2020-02-07:
Thanks for the feedback!