Heatit-Controls/no.ThermoFloor #38: Z-wave devices unreachable in Homey

JOR200, 2020-02-15:
Hi.
I have several Z-TRM2fx devices for controlling floor heating in my home. I have issues getting Homey Heating Scheduler to set all the thermostats as scheduled. Sometimes it works and sometimes not.
I have done a lot of digging to find out why this was not working and I think I found the reason.
It seems that Homey very often believes that one or more of my Z-TRM2fx devices are unreachable. This in turn messes up the z-wave routing resulting very bad / long routes.
I’m not sure how Homey handles communication with unreachable devices, but maybe it just skips sending the commands.
I think it is the Heatit z-wave driver that reports (or not reporting) the z-wave device statuses to Homey. I never had any issues controlling the thermostats from the device pane in the Homey app, so I suspect the devices are in fact communicating properly even if Homey believes they are not.

[image]

Node 2 – 8 are Z-TRM2fx devices. 9 and 10 are two different Fibaro devices. If I do a test on an unreachable device, it reports OK and the unreachable flag is removed. After a while the flag is set again. The Fibaro devices are never flagged as unreachable.
Can you please have a look into this?
I’ll be happy to provide you with more info if necessary.
Rgds.

JOR200, 2020-03-11:
Hi @TedTolboom 
Did you have the chance to have a look at this yet?

JOR200, 2020-03-24:
Hi @TedTolboom
Have you seen this issue?

JesHave, 2020-04-06:
I have exactly the same problem with my Heatit Z-wave termostat TRM2fx. Now I removed it and that is accepted but when trying to connect again the termostat and Homey reacts as intended until it is time to connect. The Homey app responds with the error message, the unit cant be reach any longer.

closed 2021-07-30 by HeatitControls