Heatit-Controls/no.ThermoFloor #39: Heat it z-trm2fx is turning off when syncing

Kongarmadillo, 2020-03-08:
Greetings,

I've tried to include the heat it z-trm2fx into Homey by using this app. The app fail to conect to the heat it thermostat. Once the pairing is made, and the first step is marked as green, the thermostat imediatly turns itself completly off, and the process fails.
Any clue how to fix this?

Kind regards Kristian

JOR200, 2020-03-09:
Just activate pairing/connect again on the thermostat without leaving the paring screen in the app. The app continues with step 2 of the pairing process. The process may end correctly or fail, but the thermostat is now connected. You need to have Homey really close to the thermostat when pairing the device.
Rgds. Jan Olav

Kongarmadillo, 2020-03-09:
Thanx! That did it :)

Kristian

closed 2020-03-09 by Kongarmadillo