Heatit-Controls/no.ThermoFloor #48: Z-TRM3 not reporting power consumption

danaksel, 2020-07-26:
Hi!
I’ve just realized that my Z-TRM3 isn’t reporting the right power consumption. Mine has reported 14 watts 24/7 for since installed several weeks ago. My friend has the exact same issue. Something that can be looked into?

vsurlien, 2020-07-28:
Same problem here. Mine rapported correct Watt once, and then steady at 0.14W. kWh is and temp are working.

vsurlien, 2020-10-28:
I tried changing association group as described here. This worked for me. 
https://community.athom.com/t/heatit-z-wave-app-v2-7-0/166/333

TedTolboom, 2020-10-28:
Are all issues occuring on Homey v4.2 or on Homey v5.0?

vsurlien, 2020-10-28:
My is 4.2

danaksel, 2020-10-28:
Mine is 4.2 as well.

TedTolboom, 2020-11-01:
I'm currently looking into the issue and are in contact with Heatit to clarify

TedTolboom, 2021-01-11:
I've just released the 2.7.1 update of the app, that should resolve a bug at Athom's side while setting the correct association groups, which resulted in that the Z-TRM3 thermostat did not have the right communication paths toward Homey.

After updating of the the app, the device should migrate to the right association groups automatically and should start reporting as expected.

I'll close this issue for now, assuming resolved. Please re-open when not.

closed 2021-01-11 by TedTolboom