Heatit-Controls/no.ThermoFloor #49: Z-TRM3

lurajon, 2020-07-31:
Thank you for your awesome work with this app.

I bought the TRM3 a couple of months ago (before the app had support for TRM3).

I have a problem with the thermostat where it continues to heat even though the target temp is reached. 

I also see that temperature thermostat is "internal" (not sure if this is a fixed value) even though I have configured to use the floor sensor on the device.

[image]

Best regards,

Jone

lurajon, 2020-07-31:
[image]

TedTolboom, 2020-11-01:
> I have a problem with the thermostat where it continues to heat even though the target temp is reached.
Typical the Z-TRM3 is controlling the heating system independently, based on the set temperatures and thermostat mode.
Are you sure that you setup the thermostat correctly; specifically regarding the modus?

TedTolboom, 2021-01-11:
I've just released the 2.7.1 update of the app, that should resolve a bug at Athom's side while setting the correct association groups, which resulted in that the Z-TRM3 thermostat did not have the right communication paths toward Homey.

After updating of the the app, the device should migrate to the right association groups automatically and should start reporting as expected.

I'll close this issue for now, assuming resolved. Please re-open when not.

closed 2021-01-11 by TedTolboom