Heatit-Controls/no.ThermoFloor #52: Frequents refresh parameters Heatit Z-TRM3

HiHaHop, 2020-10-22:
Hi Ted,

Last week I contact Willy from Robbshop, I think he conact you about this. He ask me to contact you via Github. So here I am :-) 

I have a Heatit Z-TRM3 for the floorheating in the bathroom. For good working flows (homey) the parameters need to up date frequently so the floorheating can turn on at the moment that is needed. If you can see on de screenshots some parameters don't refresh frequently, Some in seconds, some in minutes and that seems tot be ok. But Some parameters refresh in days....on that parameters I can't build a flow. 

Can you telle me how toe solve this so al parameters will refresh more frequently (let say in minutes?)?
If you need more info or something else, please contact me.

Thanks in advance! 

Greets Gerben

[image]
[image]
[image]
[image]

TedTolboom, 2020-11-01:
@HiHaHop, thanks for reporting the issue.

> For good working flows (homey) the parameters need to up date frequently so the floorheating can turn on at the moment that is needed.
Can you explain what you try to achieve? Because principally, the thermostat is controlling independently wether or not the floorheating is turned on or off... Homey can control the different setpoints used / thermostat mode etc, but control is done by the thermostat itself.

The temperature updates as shown in the UI (app) also depend on the sensor modus of the thermostat; check the manual of the thermostat to understand the differences between the modes.
If F-modus is used, typically on the floor sensor temperature is reported. If AF-modus is used, typically both temperatures are  reported.

TedTolboom, 2021-01-11:
I've just released the 2.7.1 update of the app, that should resolve a bug at Athom's side while setting the correct association groups, which resulted in that the Z-TRM3 thermostat did not have the right communication paths toward Homey.

After updating of the the app, the device should migrate to the right association groups automatically and should start reporting as expected.

I'll close this issue for now, assuming resolved. Please re-open when not.

closed 2021-01-11 by TedTolboom