Heatit-Controls/no.ThermoFloor #53: Z-TRM3 - AF mode - floor sensor not reporting

apekatt1, 2020-10-23:
Hi Ted,

Thanks for working on this excellent app! I recently bought some Z-TRM3 devices. They have a built in room sensor, in addition to a floor sensor. Hence I want to use this thermostat in the "AF" mode (internal+floor) in order to use the floor sensor as a limitiation to prevent the wooden floor getting warmer than 27 degrees celsius. But when i select the AF-mode Homey will only report temperature of the internal sensor (A). It does not report floor sensor (F). However when I change mode to 'F' (floor only), then the floor sensor will report the floor temperature. So I have confirmed that that the floor sensor is working.  The same happens on all my seven thermostats. 

Do you have the chance to have a look at this issue? I see there are some similar issues regarding z-trm3 temperature reporting, but I could not find excatly the same issue I experience, so I created a new issue for this specific matter, but they might all be related for what I know :). 

All my thermostats are on firmware 4.4.

TedTolboom, 2020-11-01:
Which Homey SW version are you currently using?

apekatt1, 2020-11-05:
> Which Homey SW version are you currently using?

Hi Ted,

I am using Homey Pro (early 2019) model.
Homey version 4.2.0
App version 6.2.0.652.

Thanks for assisting with this

commit 4459e522 (2021-01-11): Fix Z-TRM3 communication issues

Resolves #53, #52, #50, #49 and #48
closed 2021-01-11 by TedTolboom