Heatit-Controls/no.ThermoFloor #54: New manufacturer ID needed for Z-Push devices

enholm, 2020-12-13:
I was not able to connect my Z-Push 8 device to Homey. It ended up being added as a simple Z-Wave device. After looking at the current source-code I found that the list of manufacturers was missing the ID reported by my device. When adding 773 to the list of manufacturers in driver.compose.json it worked like a charm for me.
I have tested this for a Z-Push Button 8 only, but I guess this is the same issue with all HeatIT devices.

/Morten Enholm

TedTolboom, 2020-12-13:
@enholm thank you for reporting. 
I am aware that there is a batch of Z-pushbutton 8 which is delivered branded as Heatit but without the Heatit manufacturerID.

Unfortunately I can’t add this manufacturer ID to the app, since it does not belong to Heatit or the OEM.

enholm, 2020-12-13:
@TedTolboom thanks for responding. 
I see in the assigned list of zwave manufacturers this is another Norwegian company (Future Home AS at https://www.futurehome.io/) which does not seem to be related with Thermo-floor.no in any legal manner. They are still somehow related as Thermofloor are selling Futurehome products and Futurehome are listing Heatit as a supported brand.

I do understand you dont want to add arbitrary manufacturer ID's to an official app. It is however a bit bad for those non-developers out there that have this problem and the companies seem to have a business relation when you check their web-sites.

/Morten Enholm

hiato6, 2021-01-10:
> @enholm thank you for reporting.
> I am aware that there is a batch of Z-pushbutton 8 which is delivered branded as Heatit but without the Heatit manufacturerID.
> 
> Unfortunately I can’t add this manufacturer ID to the app, since it does not belong to Heatit or the OEM.

Hi Ted, I'm facing the same issue with the Z-Push 8, however I've purchased it directly from Heatit's team as I'm a authorized distributor, so I'm quite sure it's a Heatit authentic device.

Any Idea how we can get this fixed?

TedTolboom, 2021-01-10:
@hiato6 it's not a question if the device is purchased at a authorized distributor.

The devices are delivered by and branded as Heatit, but this specific batch has a (by the manufacturer) wrongly programmed firmware. I'd suggest to contact Heatit on this issue.

closed 2021-06-30 by HeatitControls

christianstre, 2021-11-13:
@HeatitControls 
Have you been able to solve this challenge?