Heatit-Controls/no.ThermoFloor #59: Z-DIN-616

Chiellus66, 2021-06-03:
Hi,
Since the latest update of Homey (rev 7.00-rc8) my outputs on the Z-DIN616 are not working anymore... I have Heat-it version 2.7.1 installed. On the Zwave function of Homey with Test function the node is reachable however no outputs is working.
Other Zwave devices are working correctly? Is there a issue with this?
Thanks in advance

bskog, 2021-06-22:
I'm having exactly the same issue on Homey 7.0.0. None of my Z-DIN 616's are working anymore and 100+ other devices works perfectly fine.

bptbest, 2021-06-28:
Same problem here. Has a solution been found?

HeatitControls, 2021-06-30:
Hi, we are working on finding a solution to this issue.

Chiellus66, 2021-07-12:
> Hi, we are working on finding a solution to this issue.

Hi, any news about this?

gvdijnsen, 2021-08-04:
Same problem here. The device is detected but the outputs are not working. No errors, just not working…

Chiellus66, 2021-09-06:
Hi Heat-it... is there any news on this??? I reported this 3 months ago and up till now no solution as far as I see.... Could you please indicate if there will be a solution for this??

bskog, 2021-09-06:
> Hi Heat-it... is there any news on this??? I reported this 3 months ago and up till now no solution as far as I see.... Could you please indicate if there will be a solution for this??

I recently found out that it seems to be fixed in Homey 7.1.1. You have to remove and add the Z-DIN from Homey and then it works again (at least for me).

gvdijnsen, 2021-09-06:
> I recently found out that it seems to be fixed in Homey 7.1.1. You have to remove and add the Z-DIN from Homey and then it works again (at least for me).

That is good news! Let's hope it works for me too. About time I can use my garden lights again ;-). I will test it this week and report back on it here!