Heatit-Controls/no.ThermoFloor #63: z-push button 8 can pair but not trigger flows in Homey

Matbj, 2021-06-24:
I'm successfully able to pair two new z-push button 8 devices with my Homey. However I'm not able to trigger any flows (test button works). I have another z-push button 8 that was paired from before and that one works perfectly. 

In the advanced settings I can see that the old (functional) one has a check mark under "Secure". The new ones however get: "S2 (Unauthenticated)".

Any clues? Is it the new Homey 7.0 version that broke it?

HeatitControls, 2021-06-30:
Hi, do these devices include as Z-push buttons or as generic Z-Wave devices?

Matbj, 2021-06-30:
Hi! I added them as Z-push buttons using the Heatit app in Homey. 

Now I've also tried a brand new z-push button 4 with same result.

Matbj, 2021-06-30:
Okay, I just now tried adding the button 4 as a Homey generic Z-wave device. I ended up with the exact same result unfortunately.

HeatitControls, 2021-07-01:
@Matbj What do you see on manufacturer ID?

Matbj, 2021-07-01:
I see: 411

[image]

HeatitControls, 2021-07-02:
You suggest it might be that it is version 7.0 that broke it. Does that mean it worked before?

Matbj, 2021-07-02:
Yes, I previously paired a z-push button 8 (~4 weeks ago). Which still works as should.

Matbj, 2021-07-10:
Closing issue as it got solved by Homey 7.0.1

closed 2021-07-10 by Matbj