Heatit-Controls/no.ThermoFloor #70: Heatit Z-Dim, Heatit Z-relay loosing connection and locks unit in On/Off State

Tomandrebj, 2021-09-19:
Hi

I have 15 Z-Dim and 1 Z-relay connected to homey.
I have huge issues with the dimmers and z-relay going in to a locked state.
This happens after 5 minutes of beeing connected to the hub.
Z-Dim get`s locked in either 100% ON or 0% off state, Physical button on the dimmer also gets locked and the dimmer needs a power reset to work again, untill it locks itself after 10 minutes.
This is also the same for the Z-Dim relay.
When the dimmers/relay gets locked, the hub can not access the nodes
[Z-Dim Logg from homey.txt](https://github.com/Heatit-Controls/no.ThermoFloor/files/7192142/Z-Dim.Logg.from.homey.txt)
.

Tomandrebj, 2021-12-16:
@NilsBohr 
Do you know if there is a fix for this comming?
I`v had a chat with Miguel and Jonas pointing to the S2 handshacking with homey to be the issue, but as it is now even with the driver disabled for S2 verification, Homey is still asking for S2 inclusion.

NilsBohr, 2021-12-16:
@Tomandrebj 
The Homey loosing connection was thought to be related to using old SDK v2.
We have now updated to SDK v3.
Can you check if loosing connection issue still exists?

In terms of the S2 inclusion. You can simply enter a 0-0-0-0-0 pin code and the device should be included unsecure if you don't want an S2 secure operation.

Tomandrebj, 2021-12-17:
It`s still the same with SDK v3, at least at here.
i have now removed 3 nodes and re-paired unsecure, i will do a 48h test to see if the connection is ok.