Heatit-Controls/no.ThermoFloor #75: Problems with direct assosiation Homey

Diddern, 2021-11-11:
Hello! After updating to Homey v7 it is impossible to add direct assosiations for switches.

Tested with two different Z-push buttons. According to the driver settings:
https://github.com/Heatit-Controls/no.ThermoFloor/blob/master/drivers/Z-push-button-2/driver.settings.compose.json

One device running firmware v1.26 and one with v1.28 (I can not find any documentation for firmwatre v1.28, that said I can not verify that  waking up and priming the device for direct assosiations have changed).

Log output from Homey after setting direct assosiations:
```
Command[1377] start: sendData

[ProcessSendData]: To node: 22 with data: 0x9f02700115db374214c7fa59cce3f9e9b628f520 and txOptions: TRANSMIT_OPTION_ACK,TRANSMIT_OPTION_AUTO_ROUTE,TRANSMIT_OPTION_EXPLORE

Command[1377] end: sendData

Node[22]: [COMMAND_CLASS_CENTRAL_SCENE] {"Sequence Number (Raw)":{"type":"Buffer","data":[56]},"Sequence Number":56,"Properties1 (Raw)":{"type":"Buffer","data":[128]},"Properties1":{"Slow Refresh":true,"Key Attributes":"Key Pressed 1 time"},"Scene Number (Raw)":{"type":"Buffer","data":[1]},"Scene Number":1}
```

HeatitControls, 2021-12-13:
Which device did you try associate to? Are they on the same security level? In Z-wave devices need the same security level to be able to associate together.

HeatitControls, 2021-12-14:
Hi, unfortunately Homey currently doesn't allow choosing the security level except for in the apps code.
I am unsure of which security version Fibaro uses but if you include the device and when it asks for a DSK code you type 0 0 0 0 0 it might work.

Diddern, 2021-12-14:
Direct assosiations does not work between S1 and S2 security. Both devices need same security level, and that is the root of the problem. 
S2 is not backwards compatible with S1, and that means that many devices will never work with HeatIt, as they will probably never get an update.